sparkfun/OpenLog #183: Multiple files creates (intermittent garbage data)

krevans, 2015-02-25:
I am trying to print to the SD card in the datalogger, and am having issues with it creating multiple files while the code is looping (and loosing some information between files). Additionally, sometimes a few lines of garbage data are output to the SD card, usually at the beginning or end of a file (can be anywhere within the files created during a single run of the code). I see this issue has been discussed previously, but I've tried to update the firmware to the latest version and this still seems to be a problem.

I am currently running at 9600 baud, printing four integer values (separated by commas) every 136 ms, with the last print including a new line (println).

Do you have any suggestions to remove this problem?

@Brodi has been helping with this issue, but we are stumped!

nseidle, 2015-02-26:
Hmm, I don't have a good solution off the top of my head. I'll need to replicate it on the bench to really dig in. Can you post some code that causes the problem?

Let's also cover the basics:

What have you got it hooked to?
What pins are hooked up?
What operating voltage?

krevans, 2015-02-28:
I think the problem may be that I am trying to send information too quickly. I am trying to send 4 integers, comma delimited, at about 200 Hz frequency. When I slow down the cycle time the problem does not seem to occur. Do you think this might be what was causing problems?

nseidle, 2015-08-20:
9600bps is 960 bytes per second. I can transmit at this speed constantly to OpenLog without it dropping a character. Depending on the extra characters you are transmitting, for example "4,6,11,5" that would be 8 characters at 200Hz. This is 1600 bytes per second, significantly faster than the 9600bps you are communicating at.

Your original comment is "four integer values (separated by commas) every 136 ms, ". This is doable "1,2,3,4," or 8 bytes at 7.35Hz. Much slower. OpenLog should be able to handle a few bytes every few hundred milliseconds.

Because of the age of this issue (sorry I'm slow!) and because it sounds like you've settled on a solution I am going to close it. If you have new information or example code please re-open.

AyrtonEstrella, 2018-08-21:
Hi @nseidle !
I know it's been 3 years since this was posted. But I recently got my own original OpenLog and I am having this same exact issue.
I'm attaching a screenshot of the text I'm trying to capture with the OpenLog.
I'm using this configuration:

- Firmware: OpenLog Minimal
- Baudrate = 9600bps

I don't know how to measure how many bytes are sent each second, but the data you can see on the screenshot takes less than 2 seconds to be printed.

Any ideas?

Thanks!
[image]

nseidle, 2018-08-27:
Hi! Welcome back. Let's solve this!

Let me get the basics out of the way so we don't waste time chasing the wrong thing:

* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

Can you post a text file of that output so that we can measure the number of bytes? With the raw text of your output we can also look for things like ctrl+z popping up. I doubt that's the issue.

Once I have a copy of that output I can replicate your work by sending the same text to a unit I've got here.

AyrtonEstrella, 2018-08-27:
Hi Nathan!

  *   I’m not sure about what you mean by “Platform”. If you mean the device to which I am connecting the OpenLog, its a 3.3V PIC18F26J50.
  *   I purchased the OpenLog a couple of weeks ago. So I assume its the latest OpenLog version.
  *   The firmware version is also the latest, downloaded from github.
  *   This are the connections using jumper wires, I am testing the Minimal version of the OpenLog firmware, and modified the baudrate to 9600:
     *   PIC - OpenLog
     *   3.3V - VCC
     *   GND - GND
     *   TX - RX
  *   The target device is powered by a external 12V power supply, followed by a 4V switching regulator and then a 3.3V LDO (on the target device).
  *   More details:
     *   I am not sure if the 3.3V input voltage is enough to provide stable 3.3V to the OpenLog, because its passing through the OpenLog’s LDO.
     *   I tried modifying some options of the config file, like ignoring the escape character but I had no luck.

Thanks!
--
Ayrton Estrella Alcocer
Móvil: (999) 207 4796

nseidle, 2018-11-03:
Your setup sounds fine. Out of curiosity, why did you go with the Minimal firmware version? The stock version should work fine for this application.

Can you post a file of what you're sending to the OpenLog so I can try to replicate?

My only guess at the moment is your gut feeling which is maybe the voltage running OpenLog is too low. If you can, check VCC on OpenLog when powering your system. Everything should work if VCC on OL is above about 3.0V.