sparkfun/OpenLog #185: 921600 Baudrate and continuous logging

Leek1001, 2015-04-02:
I would like to use this logger as a sniffer.
OpenLog v3.20
Connected to a serial output with a baudrate of 921600.
Pins used: +3v3, GND, TX, RX
The flashcard what is inserted is: 8GB FAT32 SDHC class 4.
The configuration file on this flashcard is: 
921600,26,3,1,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

While booting it flashes a couple of times green and then 3 times blue, goes of and now and then 1 times blue.

But it will not work and log all kind of stange characters. Do you have any idea how I can get this to work?

nseidle, 2015-04-02:
You are using TTL serial not RS232, correct?

I've tested with 1,000,000bps from an Arduino to OpenLog and it worked well because the ATmega on OpenLog could use a divisor that was a whole integer. I'm not sure what the serial error will be at 921,000 bps, I have never tried that before. 

If possible, try (or emulate) the system at a slower baud rate just to make sure your system is setup correctly.

thenickdude, 2015-04-02:
Some baud rates have quite inaccurate clocks, if I log at 230400, the error is 8% (from memory) and only garbage will be logged unless I set the data source to use 2 stop bits to aid synchronisation. 

The 250000 baud rate has no error and logs flawlessly.

Leek1001, 2015-04-03:
@nseidle,
I've tested it with a FTDI cable 'TTL-232R-3v3-WE' connected to the board and than logging to the computer is going okay, but when I connect the OpenLog it will not log correctly.

nseidle, 2015-04-03:
Ok so you are interfacing with TTL level. 

Off the top of my head I can't remember if Arduino sets U2Xn to 1 (I think so). So from page 173 of the ATmega328p datasheet:

UBRRn = 16,000,000 / (8*921600) - 1 = 1.17013
UBRRn = 1 (we have to round 1.17 to whole integer)
BAUD = 16,000,000 / [8*(UBRRn+1)] = 1,000,000bps

(should be) 921,600bps / (is) 1,000,000bps = 92.16 (no so great, 8% error)

Because the ATmega on OpenLog is running at 16MHz it can only really receive at 1Mbps, 666,666bps, 500,000bps, etc. I'm afraid at 921,600bps is going to have considerable receive error rates. If you can change the baud rate of the thing you are trying to log to 1Mbps it should work much better.

closed 2015-08-20 by nseidle