sparkfun/OpenLog #186: three CNTRL Z's in data file instead of SEQLOG.txt

Rande4000, 2015-04-28:
On occasion when I send the enter command mode characters they are included in my data file I specified using the append command, example file name 2015-04.dat. normally these characters go in the seqlog.txt file.
Here is an example of the error within the data file:

[image]

On at least one occasion it caused the open log to stop recording.
Have you seen this before?

aljg91, 2015-06-01:
I am facing to the same issue.
Set up openlog in command mode, then create a file with "append" command which set openlog in "data" mode and send data to file ; all is OK
But then it is impossible to drop down into command mode by sending three CTRLZ ; these are written in the data file and openlog stay in "data" mode.

nseidle, 2015-08-20:
@Rande4000 It looks as though your system is automatically sending the escape characters. Is your system sending the characters or is a human? I ask because the escape character parser is not perfect and will sometimes read (based on your example):

    ....Mph,11.5,138<cr>^z

OpenLog will reach the end of the buffer and record this set of data to SD. It will then resume listening and hear:

    ^z^z2015-04-27 14:58....

Because the buffer was split the parser didn't catch the three escape characters are actually sequential. So ya, are the escape characters being passed quickly and in line with other data? It will work and drop to command prompt sometimes but other times it will be missed.

@aljg91 From what you describe I cannot replicate your problem. Here's what I'm inputting to my OpenLog via the terminal:

    new test.txt
    append test.txt
    <[type some characters then type ctrl+z three times]~>              (~> indicates I've been dropped to command line)

Are you sure you have ctrl+z as your escape character? It is by default but can be reconfigured to any character. If it is indeed ASCII 26 (ctrl+z), are you sure you are sending ctrl+z?

aljg91, 2015-08-24:
[@aljg91]
Thank you for your replies.
I stopped working on this project during this summer and will start again next month.
Following nseidle advice I will check if my system is sendind ctrlz or not, wait some time before sending the three ctrlz and let you know if this solve the problem.

Turtle64, 2015-10-13:
Hi,
I am facing this issue now too.
My program logs data every 5 seconds. At midnight I create a new file by sending this sequence:
3 escape chars
wait 200ms
send "new 20151013.bin" (file name = date)
wait 200ms
send "append 20151013.bin"
wait 200ms

continue logging every 5s.

This has worked successfully for about 500 days. All files are still on the SD card.
But now suddenly the openlog logs the 3 escape chars into the data file and does not create a new file but continues logging in the file of the previous day. In the last month it has only created 4 new files, all other data went into "old" files from previous days.

Could this be an issue of too many files in the root directory?

nseidle, 2015-11-12:
Hi Turtle - 500 files might be the problem... The exit characters should be caught, then the new command might take some significant time with 500 files in the root directory. But it's odd that it stopped exiting all together. I'll try generating a bunch of text logs and replicate your problem.

bandigore, 2015-12-01:
I've been having this problem for a while.  I create several files and then repeatedly add to them using append.  Everything works great ... except the control characters showing up on each new append.  I've tried the delay as suggested and have even waited for the command line character to show up before appending.  I've also tried different control characters.  Nothing I do seems to get rid of the control characters in the file.  As opposed to the above post, this has been happening from the beginning and continues no matter what changes I try.  I can post the files, but they are essentially the same as Rande4000's problem.  And yes, these are microprocessor generated commands and data.

nseidle, 2015-12-02:
Hi Bandigore - We have multiple different issues on this ticket unfortunately.

I believe your issue is that control characters are being recorded to the log file. This is correct behavior for v3.x of OpenLog. If this is not your issue please create a new ticket.

Good news however: In the new v4 version of OpenLog escape characters are removed from logs.

bandigore, 2015-12-02:
Yes, that's my issue...escape characters are recorded to the files.  It
wasn't my understanding that this was 'correct' behavior.  If it is, I
guess my OL is working 'correctly'.

Is the new version a flash upgrade?  Or new hardware?

John

bandigore, 2015-12-03:
Found the firmware.  Thanks,

John

closed 2016-01-11 by nseidle