sparkfun/OpenLog #187: File modification date

pgebhard, 2015-05-17:
Is it possible to adjust the modification date generated for each new file on the SD card? As it is, mine always dates files as being modified on Jan 1, 2000.

thenickdude, 2015-05-18:
The OpenLog doesn't have a real-time clock onboard so it can't keep time for you. But are you asking if there is an API for setting the timestamp on a file?

pgebhard, 2015-05-18:
Yes, an API would be sufficient. I'm planning to get a time sync over an Xbee connection first.

nseidle, 2015-05-18:
The best way I've found is to add a [GPS module](https://www.sparkfun.com/tutorials/282) (jump down to 'Main Controller Board') or an RTC and print the current date/time to the log. I then use a python script to pull out the date from the log and rename the files. Not ideal but workable.

Allowing the end user to assign a modification date is tricky. The user would send a command, then the date/time, then OpenLog would have to track it until it was power cycled. I'm afraid we don't have the room in flash to handle this amount of overhead. I'm sorry. Please re-open or comment if you think it's a must have feature.

closed 2015-05-18 by nseidle

shadimoustafa, 2019-04-15:
Hi Nathan,
I am looking for the same feature to add a timestamp to the log file name so I can distinguish between them as I am using the OpenLog as a black box device. 
Thanks