sparkfun/OpenLog #188: Won't compile with Arduino 1.6.4

ohhorob, 2015-05-21:
Arduino: 1.6.4 (Mac OS X), Board: "Arduino Uno"

    Build options changed, rebuilding all
    In file included from OpenLog_v3.ino:180:0:
    /Applications/Arduino.app/Contents/Java/libraries/SerialPort/SerialPort.h: In member function 'size_t SerialPort<PortNumber, RxBufSize, TxBufSize>::write(const __FlashStringHelper*)':
    /Applications/Arduino.app/Contents/Java/libraries/SerialPort/SerialPort.h:583:11: error: 'prog_char' does not name a type
         const prog_char* p = (const prog_char*)s;
               ^
    /Applications/Arduino.app/Contents/Java/libraries/SerialPort/SerialPort.h:584:25: error: 'p' was not declared in this scope
         size_t n = strlen_P(p);
                             ^
    Error compiling.
    
Per instructions on this GitHub repo, I downloaded SerialPortBeta20120106.zip and unpacked it, moved the SerialPort folder to the libraries directory.

BrentWilkins, 2015-05-21:
I don't have a way to test right now, but you might try replacing `const prog_char* p` with `const PROGMEM char * p` or `const char* p`.
prog_char is depreciated in avr-libc.  You might first try:
`#define __PROG_TYPES_COMPAT__`

nseidle, 2015-08-20:
I have run into piles of problems getting OpenLog to correctly compile under Arduino 1.6.x. At this time it only works under Arduino 1.0.x. Sorry! I'll update the wiki instructions and this issue as things change.

closed 2015-08-20 by nseidle

CerealKiller1, 2015-09-19:
Hello, sorry to bring this back up (this is my first ever issue tracker of any project...)

I am trying to compile OpenLog to install it on a custom board I made. I sourced 328P's with the arduino bootloader already installed on them. The problem is I can not compile, I get a LONG LONG list of errors. I reverted back to arduino 1.0.5 and installed the two libraries needed (SerialPort and SdFAT) yet it doesnt even come close to working. I have also tried sellecting differnt board types and tried both the regular and the lite versions.

I will copy/paste the list of error messages below. I hope my intended use is ok, it complies with the CC SA v3 license and is an open source project I'm working on @ RCGroups

Please note I tried both, downloading the linked directory from the "flashing FW" wiki page as well as downloading from the repo, I also tried copying the text from the files in the repo directly into a new sketch, I've also verified the libraries are installed, nothing works. I'm use to going threw and fixing [sometimes a lot] of errors but this just seems like something's fundamentally messed up. Thanks in advance! If this should be split into it's own issue then I'm sorry...

errors:
sketch_sep19a:5: error: expected constructor, destructor, or type conversion before '<' token
sketch_sep19a:70: error: 'Sd2Card' does not name a type
sketch_sep19a:71: error: 'SdVolume' does not name a type
sketch_sep19a:72: error: 'SdFile' does not name a type
sketch_sep19a.ino: In function 'void systemError(byte)':
sketch_sep19a:103: error: 'NewSerial' was not declared in this scope
sketch_sep19a.ino: In function 'void setup()':
sketch_sep19a:147: error: 'NewSerial' was not declared in this scope
sketch_sep19a:159: error: 'card' was not declared in this scope
sketch_sep19a:159: error: 'SPI_FULL_SPEED' was not declared in this scope
sketch_sep19a:160: error: 'volume' was not declared in this scope
sketch_sep19a:160: error: 'card' was not declared in this scope
sketch_sep19a:161: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:162: error: 'volume' was not declared in this scope
sketch_sep19a.ino: In function 'char* newlog()':
sketch_sep19a:200: error: 'SdFile' was not declared in this scope
sketch_sep19a:200: error: expected `;' before 'newFile'
sketch_sep19a:225: error: 'NewSerial' was not declared in this scope
sketch_sep19a:239: error: 'newFile' was not declared in this scope
sketch_sep19a:239: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:239: error: 'O_CREAT' was not declared in this scope
sketch_sep19a:239: error: 'O_EXCL' was not declared in this scope
sketch_sep19a:239: error: 'O_WRITE' was not declared in this scope
sketch_sep19a:242: error: 'newFile' was not declared in this scope
sketch_sep19a:242: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:242: error: 'O_READ' was not declared in this scope
sketch_sep19a:256: error: 'NewSerial' was not declared in this scope
sketch_sep19a:260: error: 'newFile' was not declared in this scope
sketch_sep19a.ino: In function 'void seqlog()':
sketch_sep19a:290: error: 'SdFile' was not declared in this scope
sketch_sep19a:290: error: expected `;' before 'seqFile'
sketch_sep19a:296: error: 'seqFile' was not declared in this scope
sketch_sep19a:296: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:296: error: 'O_CREAT' was not declared in this scope
sketch_sep19a:296: error: 'O_WRITE' was not declared in this scope
sketch_sep19a:298: error: 'NewSerial' was not declared in this scope
sketch_sep19a:302: error: 'seqFile' was not declared in this scope
sketch_sep19a.ino: In function 'byte append_file(char*)':
sketch_sep19a:316: error: 'SdFile' was not declared in this scope
sketch_sep19a:316: error: expected `;' before 'workingFile'
sketch_sep19a:321: error: 'workingFile' was not declared in this scope
sketch_sep19a:321: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:321: error: 'O_CREAT' was not declared in this scope
sketch_sep19a:321: error: 'O_APPEND' was not declared in this scope
sketch_sep19a:321: error: 'O_WRITE' was not declared in this scope
sketch_sep19a:322: error: 'workingFile' was not declared in this scope
sketch_sep19a:329: error: 'NewSerial' was not declared in this scope
sketch_sep19a:349: error: 'workingFile' was not declared in this scope
sketch_sep19a:363: error: 'workingFile' was not declared in this scope
sketch_sep19a.ino: In function 'void check_emergency_reset()':
sketch_sep19a:429: error: 'card' was not declared in this scope
sketch_sep19a:430: error: 'volume' was not declared in this scope
sketch_sep19a:430: error: 'card' was not declared in this scope
sketch_sep19a:431: error: 'currentDirectory' was not declared in this scope
sketch_sep19a:432: error: 'volume' was not declared in this scope
sketch_sep19a.ino: In function 'void read_config_file()':
sketch_sep19a:549: error: 'SdFile' was not declared in this scope
sketch_sep19a:549: error: expected `;' before 'rootDirectory'
sketch_sep19a:550: error: expected `;' before 'configFile'
sketch_sep19a:551: error: 'rootDirectory' was not declared in this scope
sketch_sep19a:551: error: 'volume' was not declared in this scope
[... 4 lines cut ...]
sketch_sep19a:584: error: 'configFile' was not declared in this scope
sketch_sep19a:585: error: 'rootDirectory' was not declared in this scope
sketch_sep19a:677: error: 'NewSerial' was not declared in this scope
[... cut ...]
sketch_sep19a:780: error: 'O_WRITE' was not declared in this scope
sketch_sep19a:781: error: 'NewSerial' was not declared in this scope
sketch_sep19a:803: error: 'myFile' was not declared in this scope
sketch_sep19a:804: error: 'NewSerial' was not declared in this scope
sketch_sep19a:806: error: 'myFile' was not declared in this scope
sketch_sep19a:816: error: 'rootDirectory' was not declared in this scope

This is when pasted into a new sketch but it's the same list of errors no matter how I go about it. Please help! Thanks again.

argtrak, 2016-04-16:
I know it's frowned upon to resurrect old things, but I prefer keeping relevant stuff together instead of looking at dozens of posts for a solution that should be with the question.

I hit the same problem and was able to compile in 1.6.5 by tweaking the code a bit.  There are a few different ways we can make things work, but in essence, the compiler must see __PROG_TYPES_COMPAT__ prior to any inclusion of avr/pgmspace.h in the compile process in order for the SerialPort library to compile and be usable (in this context at least).

The easiest way I found is:
1. Add the PROG_TYPES_COMPAT line in SerialPort.h:
`#ifndef SerialPort_h`
`#define SerialPort_h`
`#define __PROG_TYPES_COMPAT__ //Needed to get SerialPort.h to work in Arduino 1.6.x`
2. Move the `#include <SerialPort.h>` statement to the top of the include block OpenLog.ino
` */`
`#include <SerialPort.h>`
`#include <SPI.h>`
`...`

This way it seems to fix both the library AND the sketch.
Hope it helps.  (Now back to trying this thing out!)

joneseym, 2021-03-28:
@argtrak it may be 5 years later but thank you, this helped me.  I'm surprised that Sparkfun have not fixed this in all that time.  The Openlog bundle referenced from the Openlog guide still contains this issue and I wasted a lot of time trying to get it to compile.

Hermesfrank, 2021-06-01:
@ohhorob's comment above (May 20, 2015)...

I had same problem today! June, 2021...the "fix", i.e.:   _#define __PROG_TYPES_COMPAT__ //Needed to get SerialPort.h to work in Arduino 1.6.x_, was already in the .ino file...

I downloaded IDE V1.6.5 for this purpose, and used it, so not sure what may still be lingering out there to cause such an issue so many years later - I guess @jonesym saw similar a few months ago...

Wonder what will happen if use current IDE?

If anyone has an insight as to another fix, please advise...

markingle, 2022-09-11:
This one was tough....I got lucky and found the solution.  Simply change the following in Serialport.h.....

const prog_char *p = (const prog_char *)s  to   PGM_P p = (PGM_P)s;

There are two lines in the SerialPort.h that need to be changed...here is a screenshot from my header file that has been updated...
[image]