sparkfun/OpenLog #191: LED not working on stm32

puedpued, 2015-07-15:
The openlog works fine with arduino, the blue LED is ON when data is being transmitted.
When I use it with an STM32F411 microcontroller, the data transmits normally but the blue LED is not lit to show if it's working or not.

nseidle, 2015-07-15:
Hmm. Do you have the ability to check the communication with a logic analyzer? Do you know what voltage the STM is communicating at?

thenickdude, 2015-07-20:
To clarify, are you saying that the data is logged fine but the blue LED doesn't light up?

puedpued, 2015-07-20:
I know what the problem was. I had the transmitting rate too slow, hence, causing the led to hardly flash and my room was too bright to actually realize it was flashing. Once I increased the trandmitting rate it all works fine :)

closed 2015-07-20 by puedpued