sparkfun/OpenLog #199: v15 hardware revision notes

lewispg228, 2015-12-01:
Hey Nate,
Rev v15 looks good. Just a little feedback:

1) The bottom side fiducial (on the bottom right corner) could be moved north to avoid your "BLK" silk. This should help it be much more legible, and PnP/AOI won't mind that fid being a bit further away from the corner.

2) During testing I noticed that the blue LED blinks quite a bit brighter than the green. Would you be interested in tweaking the values of R1 and R6 to even out the brightness?

3) If you want to go the extra mile and make a mini OSHW logo I think that would be cool to put on tplace in the upper left corner.

mgrusin, 2015-12-01:
Small request: Is there room for cramming even tiny "BLK" and "GRN" silk on the bottom of the OpenLog?
 
Rationale: I'm currently working on a class that uses the Pro Mini + OpenLog. You have to plug the OpenLog into the Pro Mini "upside down", and because neither of them have BLK/GRN silk on the back side, it's hard for a noob to match up BLK / GRN without some gymnastics. BLK / GRN silk on the back of one or both of these boards would make things just a little smoother.

Thank you!

nseidle, 2015-12-01:
@mgrusin - I agree! We added silk screen labels to bottom side (SD holder side) so you should be set.

nseidle, 2015-12-01:
@lewispg228 

1) Good catch.
2) I'll double check on my units but brightness of LEDs seems good to me.
3) Hah :) Yea. I'll give it a try.

nseidle, 2015-12-01:
Changes pushed to v15 branch. Please have a look!

closed 2015-12-01 by nseidle

lewispg228, 2015-12-01:
Looks great thanks! PS If you end up wanting to change the values on those LED resistors, that could always happen before we go to reds. Rock on.

mgrusin, 2015-12-01:
That's awesome, thanks Nate!