sparkfun/OpenLog #200: Maximum ATmega328 frequency

YvonKAP, 2016-01-07:
Hi,
I'm currently working with the OpenLog V11 board and I noticed that the crystal is 16MHz.  According to the ATmega328 datasheet, at 3.3V, 16MHz is outside the maximum frequency safe operating area.  See Figure 29-1, page 303, ATmega328 full datasheet.  Am I missing something?

http://www.atmel.com/images/atmel-8271-8-bit-avr-microcontroller-atmega48a-48pa-88a-88pa-168a-168pa-328-328p_datasheet_complete.pdf

Thanks,

BrentWilkins, 2016-01-07:
You are correct. We are running that IC out of the manufacturer's recommended safe area.

YvonKAP, 2016-01-07:
Thanks Brent.  
Is there a particular reason?  
If we use 12MHz instead, would the ATmega328 still be fast enough to do its job?

BrentWilkins, 2016-01-07:
The engineer wanted to be able to process buffers as quickly as possible. A common 12.5MHz clock would run in the safe operating area of Figure 28-1 of the datasheet. I believe any clock speed at least 8MHz and less than 50MHz will interface the µSD card at the same speed. That's how the Arduino SD library that's based on the same code (sdfatlib) works. Unfortunately last time I looked a lot of stuff was hard coded in Arduino to run at the more common 8MHz, 16MHz, etc. rates. I'd look into that before changing hardware.

nseidle, 2016-01-11:
Hi Yvon! Yep - this has been discussed in the product comments (our comment system pre-dates our use of github) [here](https://www.sparkfun.com/products/9530#comment-53bb7377757b7f305b8b4567) and [here](https://www.sparkfun.com/products/9530#comment-4eaad842757b7fd351001fe2). The decision was made many years ago to run OpenLog with an ATmega328P at 16MHz. This is outside the mfg recommended graph of voltage vs frequency. There is a need for speed as we pull characters from the serial buffer and out to the SD library. Running at 8MHz would degrade the performance of OpenLog for our users who are doing lots of recording, nonstop at 115200bps.

12MHz might be sufficient but, as Brent pointed out, would break the underlying serial libraries that Arduino is built on.

For now I'm closing the issue. Please re-open if you have any other concerns.

closed 2016-01-11 by nseidle

YvonKAP, 2016-01-12:
Thanks for the reply.  Maybe on a future version, you can power the AtMega at 5V and have 5V to 3.3V level shifter to access the card if there's room.  i know there's not much space available on this one.