sparkfun/OpenLog #203: only write every other record

s-hicks2, 2016-02-26:
I've been using more than 20 of these for the past year.  I've been very happy with their performance so far, but It seems like almost a third of the ones I bought will skip every other serial string when writing to the card.  No matter what I do, they always miss data in an alternating pattern.  

For example, I power it up via the Arduino board using a relay, give it time to read the config.txt (I currently have it appending the sequential file), then send it a serial string, give it a second, then power it off.  Five minutes later, I repeat the process.  What I find is that for some of OpenLog boards, the text files will only have data every 10 minutes.  Keeping the Arduino/relay circuit the same and just swapping out the OpenLog module for another one, then the data is correctly recorded every 5 minutes.  

So I've got 5 or 6 of them lying around that aren't being used because they aren't reliable enough.  Would reflashing with new code help, or is it a hardware problem, or something else?

mgrusin, 2016-02-29:
Hi there, sorry to hear you're having these problems. The developer will probably step in and try to help, but my first guess is that the bad ones aren't being given enough time to write the received data to SD before you power them down. The firmware first stores any received data in RAM, and only writes it to the SD when the RAM buffer is full, or if 500ms elapses since the last character was received. If the log is powered down before 500ms elapses, or if a stray character extends the timeout, you'll lose the data in RAM. This doesn't explain why some of your logs are working and some aren't though. 

Updating to the latest code couldn't hurt; at least you'll know they all have the same baseline. (You can also check the version number of each unit in the menu if you connect to it interactively.) Beyond that, my advice is to ensure you're giving them at least one solid second (1000ms) before powering them down (500ms + a few 10s of ms to allow the write to complete + safety margin) as well as ensuring that all characters have really stopped during that time period. And since these go into a low-power sleep mode after the 500ms timeout anyway, you might consider NOT cutting power to them, and just leaving them connected to power the whole time.

If none of this fixes your problem, please contact our tech support department (techsupport@sparkfun.com). They'll probably want to examine one of the failing ones. Again sorry for the problem, I hope this helps!

nseidle, 2016-06-09:
Hi s-hicks2 - Can you post a basic example sketch that demonstrates this behavior? I'm wrapping my head around the problem.

> For example, I power it up via the Arduino board using a relay, 

You control power to the OpenLog via a relay? This should be ok but OpenLog requires only a few mA so you could power it through a NPN transistor if it's easier/smaller for you.

What voltage are you providing to OpenLog via the relay? 5V should be ok.

> give it time to read the config.txt (I currently have it appending the sequential file), then send it a serial string, give it a second, then power it off. Five minutes later, I repeat the process. [...]

Very odd. I agree with mgrusin, the OpenLog will auto-record after a few hundred milliseconds if no additional serial data is received. Perhaps OpenLog is not being

nseidle, 2016-10-11:
Closing because of lack of activity. s-hicks2 if you have any updates please re-open and let us know!

closed 2016-10-11 by nseidle