sparkfun/OpenLog #205: variable types

m72m72, 2016-06-25:
1.
readConfigFile():

c must be "int", byte can not be lower than zero, so no break on eof. Also the end of config record can be detected as c is not comma or digit

    if ( **(c = configFile.read()) < 0**) break; //We've reached the end of the file
    if (!(c==',' || (c>='0' && c<='9'))) break; //Bail if we hit the end of this string // masia enable

2.
append_file()

**byte**  charsToRecord = NewSerial.read(localBuffer, sizeof(localBuffer)); //Read characters from global buffer into the local buffer

if size of localBuffer is greater than 255 (like in OpenLog) charsToRecord can be zero instead of 256.

nseidle, 2016-08-19:
Hey! Thanks for the help. The coffee has not yet kicked in yet so I'm still trying to absorb your comments

1) Yep. That's a good bug catch. I like your proposed solution as well.

2) What's the issue? Is it that j < 6 and should be j < 8? I'm wondering how this code pulls in '115200' correctly. Because it does, but it shouldn't.

3) Good find. #define LOCAL_BUFF_SIZE 256 so in the cases where localBuffer is filled to capacity, it will return 256 which will be stored as 0 in charsToRecord. I'll fix these and do some testing as soon as I can.

nseidle, 2016-10-11:
1) Fixed.

2) Code works just fine for '115200' but I increased the check on j to 7. I don't know of any using more than 115200bps but 7 will allow for up to 1mbps (at least the setting will work)

        for (j = 0 ; settingsString[i] != ',' && i < len && j < 7 ; )

3) Fixed.

closed 2016-10-11 by nseidle