sparkfun/OpenLog #206: lower power usage

bandigore, 2016-08-09:
Is there any way to optimize the power consumption of OpenLog?  I am operating on battery power over long periods and need to conserve every electron.  Compared to my other components, OpenLog is quite power hungry.

I've thought of turning the power off between log writes (about every 30 sec).  But I'm not sure if this will gain me anything with the (assumed additional) power use during initialization.

My thought is to optimize power consumption by shutting down all unused portions of the processor. I know this is possible on the newer chips, but not sure about this one.  Maybe you've already done this?

Also great would be to add a sleep mode and be able to turn off the LEDs.

nseidle, 2016-08-18:
Power consumption is already pretty low. OpenLog automatically goes to sleep when a serial char is not received for more than 500ms:

See line 454: https://github.com/sparkfun/OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino#L454

I believe current is around 1.5mA during sleep/idle and ~15mA when 100% logging.

Closing out. Let me know if you have other thoughts.

closed 2016-08-18 by nseidle

bandigore, 2016-08-19:
In the power save mode the processor should be using less than 1 ua.  Add
in the crystal and the regulator and I would think it would be possible to
achieve an OpenLog sleep mode of under 100 ua.  I need to get way under the
1.5 ma.  I haven't really looked at the code in any kind of depth yet.  If
you can point me to where the current sleep/idle is implemented, I'll try
to find some time to tinker with a deeper sleep mode.

Thanks,

John

nseidle, 2016-08-22:
Hi John, 

Code relating to powering bits down is here: https://github.com/sparkfun/OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino#L206

Also a tutorial I wrote awhile ago may be helpful to you: https://www.sparkfun.com/tutorials/309

I suspect the microSD card may be a vampire draw of unknown amounts (varying mfgs may have varying draws). I agree the regulator may be part of the draw but the quiescent current should be minimal.

Let me know if you find any places for improvement! I would love to make things better.

bandigore, 2016-08-22:
> OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino#L206

Thanks! It's a great little tool, but needs a few refinements. I'll see
what I can squeeze in in between my other commitments.

John

bandigore, 2016-08-23:
I think what I'll try first is to completely shut down between writes. What
would be the consequences of turning power on and off repeatedly (say every
few minutes)?

John

On Aug 22, 2016 7:58 PM, "John der Kinderen" <johndk@wnwn-llc.com> wrote:

Thanks! It's a great little tool, but needs a few refinements. I'll see
what I can squeeze in in between my other commitments.

John

nseidle, 2016-08-23:
The unit already shuts down if no characters are received after 500ms, so no consequences? :) Please have at it and let us know!

bandigore, 2016-08-23:
I'll let you know how it goes. I don't mean an idle state, but a no-power
state. It seems that should be more energy efficient if it doesn't cause
side effects.

John

lipopo, 2018-01-04:
So are there any solution that can make much more power save now?