sparkfun/OpenLog #207: FAT16 to many files

damschneider, 2016-10-21:
hi,
we run Version 3.1.3 from OpenLog and our Problem is that we have around 500 LOGxxxxx.TXT files in the main Folder.
if we reached so many files we can`t use the OpenLog. If we remove some files we can again. sometimes later we have 500 Fails again and the OpenLog stopped working again.
in this Situation, we can`t go in command mode.
is there a solution to re activate the OpenLog WITHOUT removing the SD Card?

nseidle, 2016-10-24:
Hi,

Depending on your application, you could change to Sequential Log mode:

https://learn.sparkfun.com/tutorials/openlog-hookup-guide#configuration-file

Instead of creating a new log every power-cycle your system would add to the one log file.

damschneider, 2016-10-24:
hi,
if i change 57600,26,3,0,1,1 to 57600,26,3,2,1,1 the Controller start`s next time in command mode. But i at the Moment i cant Change the config.txt becouse i cant go in command mode. if i remove the sdcard, put it in my pc and Change the file and delete some LOG.txt i think it works, but how can i Change the config.txt with my spamed SDcard without removing the Card?

nseidle, 2016-10-24:
There's two ways to change the config file: Via a Computer and an SD card reader or via the triple Ctrl+Z to get to the command prompt on the OpenLog. If you can't remove the SD card and you can't get into command mode you're in a bit of a catch 22; there's no other way.

Why can't get into command mode?

damschneider, 2016-10-24:
if the sdcard is not full and i send "1A1A1A" the Controller answers and i can do erverthing. but if i have too many files in the rood Folder and the Controller can`t create the "Startup Logfile" i get no Respons on the "1A1A 1A" command...
and yes is no Option for my Situation to put the Card out of the box.

nseidle, 2016-10-24:
Ah, so OpenLog can't boot because there's too many files in the main directory, so you can't get into command mode. That makes sense. And you can't get access to the SD card to clear the files. Unfortunately I don't have a solution for you at this moment, I'm sorry. 

The future solution would be to find an SD library that supports larger number of files in the root directory.

damschneider, 2016-10-24:
it is not nice to hear this answer, but thanks for your support

nseidle, 2016-10-24:
Sorry I couldn't help more. Please open another issue for any other help.

closed 2016-10-24 by nseidle