sparkfun/OpenLog #211: Garbage Characters

dpstring, 2017-05-25:
I have an OpenLOG that is hooked up to an AliCAT mass flow controller via an 8-pin mini-DIN connector. My problem is that when it is powered up and collecting data, I get data that looks like this: 

"ßÿû %ìûë %dÿë¡!dÿû BIïÿ (d{÷ÿ (auûÿAIB}ï(d{»ÿ‰(eÝûÿ%õÿ÷ $huï "hw÷ÿ AI»¯ÿ $2ûk Hdë÷ÿ
­û{%Ñû» Ûûû úý{* 

Just a bunch of unintelligible characters. I'm curious if anyone has had this same problem. It has been suggested that the OpenLOG needs to be programmed to read and interpret/parse the data correctly. Do you all agree, what could be my problem? Sorry if this is a dumb question, I'm new to most of this.

Thanks 
Danny

nseidle, 2017-05-25:
Hi Danny - 

There could be a few culprits. What is the baud rate that the AliCAT is outputing? OpenLog needs to be configured to communicate at that same baud rate. Is AliCAT outputting TTL serial or RS232? OpenLog only works with TTL (0 to 5V) signals, not RS232 (-9 to +9V).

dpstring, 2017-05-25:
Hmm, the baud rates are both 9600 so that is good. But I believe that the AliCAT is based around an RS232 serial. If that is the case, any suggestions? Thanks for your response!

Danny

nseidle, 2017-05-25:
No problem. First, checkout the [RS232 tutorial](https://www.sparkfun.com/tutorials/215) if you haven't already. If AliCAT is indeed outputting RS232 then I would use [this](https://www.sparkfun.com/products/449) or [this](https://www.sparkfun.com/products/8780) to convert the RS232 to TTL.

closed 2017-05-25 by nseidle