sparkfun/OpenLog #212: Blue LED pulse - potential lost data

andydswing, 2017-06-11:
I'm serial writing about 15 characters every 10 seconds at 9600 baud. I have a dozen or so loggers. 

All my old loggers pulse the blue LED for about 250ms and I have had no data issues on 4GB and 8GB sandisk cards. For months, sometimes years, at a time. 

My new loggers seem to do 250ms pulses and then randomly drop to 25ms pulses and I'm not sure if they are logging data in the same reliably way. I've had a few 0 byte log files. It's a bit unnerving as these need to go into the field soon. They are right on top of a promini.. close proximity. 

Andy ideas??

andydswing, 2017-06-11:
I should mention that my ProMini is 5V and I think the Tx is also 5V. Openlog appears to require 3.3V - I never noticed this before. Is openlog compatible with 5V ProMinis?

nseidle, 2017-06-12:
It may be different versions of firmware. If you have access to a good and bad units please drop to the command prompt (usually 3 x ctrl+z) and send the '?' [command](https://learn.sparkfun.com/tutorials/openlog-hookup-guide#command-set). This will give you the firmware version on the units.

The LEDs may be behaving differently but can you confirm that data is being recorded differently?

andydswing, 2017-06-13:
The units are brand new from digikey. 

The good news is that the data seems to be fine. But it's unnerving when it goes from a relatively long blue pulse to a short blip. Something is changing.

nseidle, 2017-06-13:
It's still unclear to me if you have units of all the same age (firmware would be the same) or if you have units from very different time periods (we have customers with 4+ years in between versions).

Sounds like the units are working as designed. On [this line](https://github.com/sparkfun/OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino#L449) the write time may vary depending on the card write times and even what section of the card is being written to (it may take a few extra ms for the controller to locate a new/good section of flash for these bytes to be written to). 

I'm going to close out this issue but do please raise a flag if you see any problems.

closed 2017-06-13 by nseidle