sparkfun/OpenLog #214: Using hardware flow control

tostasmistas, 2017-06-27:
Hi,

I'm using the OpenLog to record data at 115200bps, sent by a device at a rate of 8 bytes per 1ms. The uSD card is a 32GB SanDisk Extreme PRO, UHS-I Class 3.

To have better control over the UART transmission we are using hardware flow control. For this, we shunted one of the GPIO pins of the MCU of the OpenLog to the CTS pin, and set it as a digital output that we are abe to control. In terms of firmware, we have based our work on v4.0 of the OpenLog Minimal version.

We have adjusted the loop of the `appendFile()` function so that it works as follows:

```
while (1) {
  charsToRecord = NewSerial.read(buff, sizeof(buff));
  if (charsToRecord > 0) {
    charsCount += charsToRecord;
    toggleLED(stat1); 
    workingFile.write(buff, charsToRecord);
  }
  if (charsCount >= CHAR_COUNT) {
   digitalWrite(cts, HIGH);
   _delay_ms(1);
   workingFile.sync(); // sync the card
   digitalWrite(cts, LOW);
   charsCount = 0;
  }
}
```
Instead of controlling the sync by time, we are doing it by counting the bytes. Whenever we have recorded `CHAR_COUNT` bytes (set at the moment to 384), the OpenLog should sync the card. However, before doing this, we set the CTS pin HIGH so we can suspend the UART transmission from the device, with the goal of possibly avoiding data loss. This device is capable of internally storing ~1400 bytes, which should give us time to sync the file and resume the data transmission, once the CTS pin is set to LOW.

Even though we are using this flow control technique we are still experiencing some data loss. To better understand what is happening, we recorded to the SD card the state of the CTS pin. The next picture shows the temporal representation of the state of the pin for a time when there was data loss.

[image]

For some reason, the sync operation seems to have taken longer time to complete. On average, this operation lasts ~5ms, but we registered a maximum of ~145ms, which caused the data loss. Furthermore, this occasional longer sync appears to be periodic, for instance we recorded continuously for ~35mins, and the fails happened at minutes 1, 15 and 30. 

Is the card doing some internal housekeeping that could justify this behaviour? Our goal is to record continuously at 115200bps, without data loss. 

I should also mention that because in our application we are doing some JSON encoding and decoding, we have to use the ArduinoJson library, limiting the overall available memory. As such, the serial RX buffer has 512 bytes instead of 1024.

alexpirine, 2018-01-24:
Hi,

I am also interested in the continuous 115200 bps logging. Any success since you opened the issue?

tostasmistas, 2018-01-24:
Hi,

Yes, I was able to do it successfully. I ended up removing the JSON enconding and decoding, so the serial RX buffer could have a size of 1024 bytes. This change along with the hardware flow control mechanism provided good results. 

I did extensive performance tests and I could log data for 24h-40h without loss of packets. You can see the hardware modifications I did on the OpenLog board and the firmware in this [repo](https://github.com/BITalinoWorld/firmware-loggerbit).

closed 2018-01-24 by tostasmistas