sparkfun/OpenLog #215: openlog power consumption

pvfpvf, 2017-07-04:
I hooked up an openlog to my PC. 5V supply. Nothing connected to pins 1 (marked BLK) and 6 (marked GRN). It works fine. However the current consumption is weird.

After power on (in mode 0): 21.5 mA, lowering momentarily to 6 mA whenever a character is sent to openlog
After switching to command mode: 26.2 mA continuous (stays like that after a READ or DISK command)
After giving LS command: 7 mA
After giving ECHO OFF  command: back to 27 mA (echo was off anyway)
After giving APPEND command: 1.5 mA until first char sent, then 21.5 mA continuous falling to 7 mA as each char sent.

There doesn't seem go be a command to find firmware version. I am not using an Arduino.

Is this how it should behave?

pvfpvf, 2017-07-22:
I am disappointed no one has decided to look into this matter. The current consumption is excessive and clearly irrational. There must be something wrong with this software. It limits the utility of openlog in battery powered applications.

nseidle, 2018-02-09:
I apologize for the slow response. I was incorrectly filtering github issue notifications.

Pressing ? at the command prompt will give you a list of commands as well as the firmware version at the start of the help menu. This is documented under the low level commands on the [hookup guide](https://learn.sparkfun.com/tutorials/openlog-hookup-guide#command-set).

I believe OpenLog is performing as designed. When the device is in command mode it is not low power, it does not go to sleep, it does not turn off the LEDs. As you type commands the status LED will turn on/off and the LED is responsible for ~20mA of current. The processor not sleeping is another 5-10mA.

To test overall system current consumption make sure the device is in mode 0 (config file setting). With no other wires or devices attached, insert a microSD card and power with 5V and GND. Interrupt the 5V line with a DMM in current measuring mode. When the DMM is attached the OpenLog should power up, briefly blink the LEDs, then turn off the LEDs indicating it has gone to sleep. This should draw less then 2mA (1.5mA on the unit on my desk). When you enter command mode it will increase to ~20mA as you were seeing and alternate as the status LEDs turn on/off with each key stroke received.

nseidle, 2018-04-18:
Please open a new issue. The original issue possibly describes connecting an OpenLog to a PC's serial or USB port. Both of which are generally bad ideas (12V on serial port can fry the OpenLog pins, USB is not TTL serial).

pvfpvf, 2018-04-18:
No it doesn't, and that's just an excuse toavoid facing the issue. Naturally I used an FT232R. However, I lost interest in this matter some time ago.

nseidle, 2018-04-18:
@pvfpvf - I had not heard back from you so I closed this issue. Can you answer the questions, mainly what firmware version you have?

Additionally, so I can replicate your issue:

* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

I have tried to replicate this issue but cannot. If you've lost interest, I totally understand. Perhaps @zmeeaag can help me figure out the problem.

zmeeag, 2018-04-20:
Platform: Windows/ STM32
Product/FW version: It's been a while since I've looked at this. But I believe it was the latest OL firmware. All boards were purchased across the past year. 
Connection/p: FTDI USB->TTL adapter / STM32L4 w/ HW UART
Power: Regulated 3.3V and 5V rails
Not sure it helps but the original support email went in on May 30th, 2017 at 1:44pm MDT. I can forward you the email chain if you'd like.

nseidle, 2018-04-25:
Here's what I found:

* Firmware version: OpenLog v4.0 (fresh off the shelf production unit)
* Connection: Arduino hardware TX, VCC, and GND to OpenLog
* Powered from 3.3V on Arduino (in an attempt to match your STM32 3.3V power)
* SD Type: Cheap card, 512mb
* During standby - 2.4mA
* During constant record - 7.5mA
* Sitting at command prompt waiting for character - 7.2mA

To measure the current I interrupted the 3.3V line to the OpenLog.

[image]

Above - standby

[image]

Above - constant read

[image]

Above - sitting at command prompt with LED on

I cannot replicate @pvfpvf's findings. Do either of you have simplified code that I can use to replicate your measurements? If you think this is still an issue I'm happy to do a hardware swap.

nseidle, 2018-04-25:
Btw - I added an [example sketch](https://github.com/sparkfun/OpenLog/blob/master/firmware/Arduino_Examples/Example6_GetVersionNumber/Example6_GetVersionNumber.ino) to help you obtain your firmware version without a USB connection.

zmeeag, 2018-04-29:
How about a simplified HW setup? 
I connected one to 3.3V through a multimeter and it immediately started drawing 20mA immediately on power up.

[image]
I then went to check the FW version by hooking up a 3.3V ftdi cable and putting the OL into command mode. When I did this the current increased to 23.6mA. Here is a screenshot of the serial terminal including the card information.
[image]

If I remove the card then the current draw goes down to 7mA in command mode, which is about what I'd expect (and what you've reported)

Now to the interesting bit. Some SD cards work a lot better than others.  I have an 8GB ATP card here:
Card type: SDHC
Manufacturer ID: 9
OEM ID: AP
Product: AF UD
Version: 0.2
Serial number: 1741636114
Manufacturing date: 3/2017
Card Size: 8039 MB
that does properly return down to a lower current (<2mA!) which is awesome

I have a 16GB Kensington card here:
Card type: SDHC
Manufacturer ID: 27
OEM ID: PH
Product: SD16G
Version: 3.0
Serial number: 784587265
Manufacturing date: 7/2017
Card Size: 15590 MB
That starts low (~3mA) but the second anything is written to it the current jumps to ~17mA and won't come back down until a read operation occurs (I generally use ls as it's really simple to do)

lastly I have a 16GB card I purchased from SF with the OL a while back:
Card type: SDHC
Manufacturer ID: 2
OEM ID: TM
Product: SA16G
Version: 2.4
Serial number: 2813862936
Manufacturing date: 5/2016
Card Size: 15489 MB
This one starts at ~20mA and blips *down* to ~7.5mA with each character it gets before returning up to 20mA on an idle serial line.  

Is there some "sleep" step that needs to be done to higher performance cards that got missed?

nseidle, 2018-04-30:
Really interesting. Thanks for the numbers! In addition (someday), please measure the current outside of command mode as well, as command mode is the highest (no sleep) mode. 

The current consumption of the SD card is the result of the flash controller and AFAIK sleep control is not accessible via SPI commands.

I'll keep trying to find SD cards that mimic your behavior. This issue looks like it's outside the purview of the OpenLog firmware.

pvfpvf, 2018-04-30:
Thanks to you both. You have inspired me to revisit the issue. I bought 3 openlogs in June 2017, tested one, and reported the above anomalies. I then built one into some equipment, destroyed one, and I've just managed to find the third. Meanwhile I had used it as an Arduino Uno and overwritten the openlog software. So I reinstalled the openlog bundle I'd downloaded in March 2018. Testing this with a Sandisk Ultra 16GB HC I card, it does not show the anomalies I reported. Nor is it's consumption as high. But I don't know if the SD card is the same as last June's, nor what openlog version was in the June hardware. For details of current test results, please see
https://www.dropbox.com/s/ivs1sz8oz401z8g/openlog_consumption_201804.mp4?dl=1
(about 75 MB).
In a sense I am sorry to have bothered you now about it, but it really was there last June.

nseidle, 2018-04-30:
Hi pvfpvf - I have never had such a pleasant interaction on Github! You have no need to apologize, I should have responded to your issue *much* faster than I did.

Thank you, your video is superb and highlights the current consumption in different modes.

When OpenLog is in command mode the ATmega never goes to sleep. Thus we see the ~10mA current consumption. For each letter that is pressed in command mode the LED toggles on/off. Your video expertly shows 9.2mA when the LED is off and 9.9mA when the status LED is on. Not much of a difference but it is part of the equation.

When in normal logging mode the ATmega will go to sleep when there is 500ms of no serial traffic and the unit should go down to the 3 to 4mA range *and* turn off the power LED. What I think the issue is, and your issue is valid, is that different SD cards can use different currents, regardless of how much we sleep the ATmega.

I'll grab a microSD card from our production stock to test as well because zmeeag reported high current from a card they bought from us.

And if you managed to fry an OpenLog, I'm impressed! They are pretty hardy. I will email you separately about replacing hardware.

zmeeag, 2018-04-30:
Little bit of research is pointing me towards some cards being less tolerant of floating pins than others when trying to go to sleep (See Cave Pearl Project). This sorta matches the behavior I'm seeing with the card where the power consumption drops when its active.  Might be worth looking into in the long run, but for now I've moved these over to a switched power rail. Wish I had the time to dig into this further it seems like a really fun project.

nseidle, 2018-04-30:
Hmm. Floating pins could be the problem? That's interesting. I could drive the pins on the SPI interface low before going to sleep...

Here is a [version of OpenLog](https://github.com/sparkfun/OpenLog/files/1962365/OpenLog.zip) where I drive the SPI pins low before sleep (pins 10, 11, 12, 13). I'll call it beta for now.

Current usage: v4.0 firmware / beta firmware
Transcend 8GB: 2.0/2.1 
512MB Generic: 2.4/2.4 
16GB Kingston: 3.2/3.3

I don't see much of a difference but I haven't found a power-hog SD card yet. If you can load new firmware please give it a try.

pvfpvf - I pulled a [16GB Kingston](https://www.sparkfun.com/products/13833) from our stock to check if the cards we are (currently) selling are a problem. I don't think they are.

nseidle, 2018-11-03:
Closing issue due to lack of activity. Please re-open if anyone has power issues or information.

closed 2018-11-03 by nseidle

nseidle, 2019-01-24:
This is a very new issue that is unrelated to power consumption. Please open a new issue.