sparkfun/OpenLog #216: Issue on writing to a file

xristod, 2017-09-22:
Dear All,
I need your advice on a problem I have when I am deleting and creating a file writing an incremented number into it. If you check my code below, when I press "c" the file "myLog.txt" is deleted and a new one is created writing an incremented number into it. Up to number 9 everything works fine but when it is writing number 10 then  it freezes ....

Delete file ...
File deleted!
File created!
Enter command mode!
Read of new file: 

8
>
Delete file ...
File deleted!
File created!
Enter command mode!
Read of new file: 

9
>
Delete file ...
File deleted!
File created!
Enter command mode!

When I opened the file from the PC and  checked the number written I noticed that it was not just "10" but "10".
In addition I would like to let you know that the method I have implemented for using the command "write File OFFSET" is not working and I do not know what I have done wrong.
Finally could you please let me know why each time I create a file I have to use "gotoCommandMode();" since I have declared it in the "setup()"?
Best,
George

The code I have used is the following with some modifications of the original one provided by Sparkfun....

#include <SoftwareSerial.h>

//-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
//Connect TXO of OpenLog to pin 3, RXI to pin 2
SoftwareSerial OpenLog(3, 2); //Soft RX on 3, Soft TX out on 2
//SoftwareSerial(rxPin, txPin)

byte resetOpenLog = A1; //This pin resets OpenLog. Connect pin 4 to pin GRN on OpenLog.

int number = 0;
String filename = "myLog.txt";

void setup() {

  Serial.begin(9600);

  setupOpenLog(); //Resets logger and waits for the '<' I'm alive character
  Serial.println("OpenLog online");
  gotoCommandMode(); //Puts OpenLog in command mode

  createFile(filename);
  OpenLog.print(String(number));
  gotoCommandMode(); //Puts OpenLog in command mode

  String str = readFile(filename);
  Serial.println("Read of new file: ");
  Serial.println(str);

}

void loop() {

  if (Serial.available())
  {
    char c  = Serial.read();

    if (c == 'c')
    {
      ++number;

      Serial.println("Delete file ...");
      deleteFile(filename);
      Serial.println("File deleted!");
      createFile(filename);
      Serial.println("File created!");
      //  writeFileOffset(filename, 0);

      OpenLog.print(String(number));
      
      Serial.println("Enter command mode!");
      gotoCommandMode(); //Puts OpenLog in command mode

      Serial.println("Read of new file: ");
      String str = readFile(filename);
      Serial.println(str);

    }
  }
}

//Setups up the software serial, resets OpenLog so we know what state it's in, and waits
//for OpenLog to come online and report '<' that it is ready to receive characters to record
void setupOpenLog(void) {
  pinMode(resetOpenLog, OUTPUT);
  OpenLog.begin(9600);

  //Reset OpenLog
  digitalWrite(resetOpenLog, LOW);
  delay(100);
  digitalWrite(resetOpenLog, HIGH);

  //Wait for OpenLog to respond with '<' to indicate it is alive and recording to a file
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '<') break;
  }
}

//This function creates a given file and then opens it in append mode (ready to record characters to the file)
//Then returns to listening mode
void createFile(String fileName) {

  //Old way
  OpenLog.print("new ");
  OpenLog.print(fileName);
  OpenLog.write(13); //This is \r

  //New way
  //OpenLog.print("new ");
  //OpenLog.println(filename); //regular println works with OpenLog v2.51 and above

  //Wait for OpenLog to return to waiting for a command
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '>') break;
  }

  OpenLog.print("append ");
  OpenLog.print(fileName);
  OpenLog.write(13); //This is \r

  //Wait for OpenLog to indicate file is open and ready for writing
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '<') break;
  }

  //OpenLog is now waiting for characters and will record them to the new file
}

void writeFileOffset(String filename, int offset) {

  OpenLog.print("write ");
  OpenLog.print(filename);
  OpenLog.print(" " + String(offset));
  OpenLog.write(13); //This is \r

  //Wait for OpenLog to indicate file is open and ready for writing
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '<') break;
  }

  //OpenLog is now waiting for characters and will record them to the new file
}

void deleteFile(String fileName)
{
  OpenLog.print("rm ");
  OpenLog.print(fileName);
  OpenLog.write(13); //This is \r

  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '>') break;
  }
}

//Reads the contents of a given file and dumps it to the serial terminal
//This function assumes the OpenLog is in command mode
String readFile(String fileName) {

  //Old way
  OpenLog.print("read ");
  OpenLog.print(fileName);
  OpenLog.write(13); //This is \r

  //New way
  //OpenLog.print("read ");
  //OpenLog.println(filename); //regular println works with OpenLog v2.51 and above

  //The OpenLog echos the commands we send it by default so we have 'read log823.txt\r' sitting
  //in the RX buffer. Let's try to not print this.
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '\r') break;
  }

  //This will listen for characters coming from OpenLog and print them to the terminal
  //This relies heavily on the SoftSerial buffer not overrunning. This will probably not work
  //above 38400bps.
  //This loop will stop listening after 1 second of no characters received
  String fileData = "";
  for (int timeOut = 0 ; timeOut < 1000 ; timeOut++) {
    while (OpenLog.available()) {
      while (OpenLog.available()) {
        fileData += (char)OpenLog.read();
        // if(openLogResponse.length() > 98) break;
      }

      timeOut = 0;
    }

    delay(1);
  }

  return fileData;
  //Serial.print(openLogResponse); //Take the string from OpenLog and push it to the Arduino terminal

  //This is not perfect. The above loop will print the '.'s from the log file. These are the two escape characters
  //recorded before the third escape character is seen.
  //It will also print the '>' character. This is the OpenLog telling us it is done reading the file.

  //This function leaves OpenLog in command mode
}

//Check the stats of the SD card via 'disk' command
//This function assumes the OpenLog is in command mode
void readDisk() {

  //Old way
  OpenLog.print("disk");
  OpenLog.write(13); //This is \r

  //New way
  //OpenLog.print("read ");
  //OpenLog.println(filename); //regular println works with OpenLog v2.51 and above

  //The OpenLog echos the commands we send it by default so we have 'disk\r' sitting
  //in the RX buffer. Let's try to not print this.
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '\r') break;
  }

  //This will listen for characters coming from OpenLog and print them to the terminal
  //This relies heavily on the SoftSerial buffer not overrunning. This will probably not work
  //above 38400bps.
  //This loop will stop listening after 1 second of no characters received
  String diskData = "";
  for (int timeOut = 0 ; timeOut < 1000 ; timeOut++) {
    while (OpenLog.available()) {

      while (OpenLog.available()) {
        diskData += (char)OpenLog.read();
      }

      timeOut = 0;
    }

    delay(1);
  }

  return diskData;
  // Serial.print(openLogResponse); //Take the string from OpenLog and push it to the Arduino terminal

  //This is not perfect. The above loop will print the '.'s from the log file. These are the two escape characters
  //recorded before the third escape character is seen.
  //It will also print the '>' character. This is the OpenLog telling us it is done reading the file.

  //This function leaves OpenLog in command mode
}

//This function pushes OpenLog into command mode
void gotoCommandMode(void) {
  //Send three control z to enter OpenLog command mode
  //Works with Arduino v1.0
  OpenLog.write(26);
  OpenLog.write(26);
  OpenLog.write(26);

  //Wait for OpenLog to respond with '>' to indicate we are in command mode
  while (1) {
    if (OpenLog.available())
      if (OpenLog.read() == '>') break;
  }
}

nseidle, 2018-02-08:
What version of OpenLog firmware are you using? See the [command set](https://learn.sparkfun.com/tutorials/openlog-hookup-guide#command-set). Once the device is in command mode send the ? to get the list of commands and the firmware version.

> When I opened the file from the PC and checked the number written I noticed that it was not just "10" but "10���".

The ???s are the escape characters that are sent to the OpenLog to get OpenLog into command mode.

> In addition I would like to let you know that the method I have implemented for using the command "write File OFFSET" is not working and I do not know what I have done wrong.

Not sure. I'll try to replicate it.

> Finally could you please let me know why each time I create a file I have to use "gotoCommandMode();" since I have declared it in the "setup()"?

I don't understand the question. You don't declare functions in setup(). However, you can run them. gotoCommandMode() is a function to get OpenLog out of logging mode and into Command Mode so that you can do things like create/delete/write to files.

nseidle, 2018-02-09:
I have re-written [the examples](https://github.com/sparkfun/OpenLog/tree/master/firmware/Arduino_Examples) and have written an example demonstrating how to use an offset such as

    write fileName.txt 10

Please see [example 5](https://github.com/sparkfun/OpenLog/blob/master/firmware/Arduino_Examples/Example5_WriteOffset/Example5_WriteOffset.ino).

closed 2018-02-09 by nseidle