sparkfun/OpenLog #217: RXI pin voltage 3.2v , seemingly preventing serial data reception (or it's dead)

dmalnati, 2017-11-01:
Hi,

I think my OpenLog may have gone bad (though I don't know why it was in a box for a year since I last used it successfully).

Short story is that OpenLog doesn't appear to be receiving Serial data properly.
Looking more closely with a scope (Saleae) I measure the RXI pin as having a voltage of around 3.2v, which I believe is either the cause of Serial data not being received, or a symptom of the unit having died somehow.

To illustrate I have two screenshots below:
- the first in isolation from all other circuit elements (and showing the RXI voltage)
- the second being in circuit with a microcontroller and the voltage changes on RXI as Serial data is received.  Here the voltage varies, but not full 0v-5v swings, instead it's within 2v-3.5v, which is wrong.

I have tried tying RXI to ground to reset any bad configuration (no change).
I have not flashed any new firmware because I just suspect this unit is busted.

Can you tell me what might be going on, or perhaps send me another?  My original order number was 2839165.

Again, this used to work in 2016 when I last used it.  Now it doesn't, and I didn't change anything.  I even ran the original working test code I wrote for it and no luck.

When completely isolated from any microcontroller, simply plugged into a power supply:
[image]

When on the same circuit as a properly functioning microcontroller and receiving serial data:
[image]

Thanks.

doug

nseidle, 2018-02-04:
Hi dmalnati - Sorry for the very late response. My github notifications were being filtered.

I'm happy to get you a replacement unit if you're still working on a project that could use an OpenLog.

A few things to check:

* With nothing but VCC/GND attached, do you see the status LEDs do anything? You should see activity at power on. If you don't, there's something more wrong than the RX pin.
* Format the SD card, reinsert, apply power. You should see status LEDs active then turn off. Check contents of SD card with computer. Does the config.txt file appear? If not, there may be a problem with the microSD socket.

dmalnati, 2018-02-08:
Hi Nathan,

Thank you for getting back to me.

Unfortunately I have already thrown away the module and replaced it with
another so I can't do any testing beyond what I wrote up in the bug report.

I'd be happy to take a replacement if it's covered by warranty or
anything.  I can't imagine why it died, and the replacement worked in the
same conditions that the replaced one didn't.

Thanks.

doug

nseidle, 2018-02-08:
Hi Doug - No worries. Yep, I'll email you.

closed 2018-02-08 by nseidle