sparkfun/OpenLog #218: Reduce clock speed to 8 MHz, can not properly log.

ewan1983, 2017-11-20:
I use Openlog to connect with my board, and got a lot of small files, seems Openlog have been keep resetting. I suspect it was a power issue as my board have only 3V supply, and it might not be enough to drive 16MHz clock as in default Openlog. 

So I modified the F_CPU configuration in boards.txt to 8MHz `uno.build.f_cpu=8000000L, `
and in `setup(void)` function of OpenLog_Minimal firmware, I add   
`clock_prescale_set(clock_div_2);`

The result is that I can only correctly log one character, if send more than one character only the first on is correctly logged, and the rest are corrupted. The problem is probably in serial rx part, but I haven't figure out what exactly the problem is.

alexpirine, 2018-01-24:
Hi,

Have you solved the issue?

marshalltaylorSFE, 2018-01-24:
I ran into a similar issue on another project using an 8MHz processor and 115200.  The problem is that there simply isn't enough time to service the ISR between bytes.  By the time the serial ISR completes, we're mid-way into the next byte.  Thus garbage.

If you can cause your serial transmitter to emit two stop bits instead of one, it gives an extra symbol's worth of time between the bytes.  With my test, this was enough to get the data correctly.  Otherwise, get the voltage up so you can run at 16MHz.

closed 2018-01-24 by marshalltaylorSFE

alexpirine, 2018-01-24:
Which voltage is required for 16 MHz? I currently use 3.3 V and sometimes it works, sometimes I get plenty of log files and lose data in between...

alexpirine, 2018-01-24:
Is it possible to set an arbitrary clock speed like 12 MHz?

marshalltaylorSFE, 2018-01-24:
3.3V should be fine.  ewan1983's issue is exactly a baud related problem... if you're getting solid chunks of data, then just missing some, that sounds like a different issue entirely.  Can you post another issue with more information about your setup that causes intermittent lost of data?

I explored 12MHz for my other project.  There's no easy way to scale things to 1.5x, so it would take a bit of research and modification to the project and maybe even the board files in Arduino.

JensGrabner, 2018-01-24:
Here is a calculator for that: http://wormfood.net/avrbaudcalc.php?bitrate=57.6k%2C76.8k%2C115.2k&clock=8%2C+12%2C+16&databits=8
I use 3V3 with 12MHz Clock a Baud Rate 115.2k (U2Xn = 1)  is good ...
and 3V3 with 8MHz Clock a Baud Rate 76.8k (U2Xn = 1)  is also good.

alexpirine, 2018-01-24:
Thanks for the tips. I am running at 3.3V and 115200 baud rate with the "minimal" firmware.

And I get multiple open log files instead of just one.

I will be happy to open a new issue if this not related to the clock (which is at 16 MHz by default from my understanding) / voltage.