sparkfun/OpenLog #220: Garbled Data, No Serial RX

BigDave365, 2018-07-11:
### Garbled Data, No Serial 

I cannot seem to get my Open Log to work properly.

Lights work as described in wiki. Will create a config file on sd card. Will record data file on sd card.

Data fed from GPS is totally garbled. GPS data tested good via FTDI and terminal (9600,8,N,1).

Data is also corrupted if sent directly to OpenLog via FTDI.

Openlog will NOT generate a prompt, or go in command mode.

Openlog will not reflash via Arduino IDE.

I reflashed via USBASP with clips, same result.

Any Ideas?

nseidle, 2018-07-11:
Hmm, did you use a cross over board to align the tx pin of openlog to the
rx pin of your terminal?

BigDave365, 2018-07-11:
You mean OpenLog TX to FTDI RX and vice versa? Yes, I check it backwards as well, and the Openlog receives no data.

Is there any way to get an older copy of the firmware to give that a try?

Thanks for the help!

-Dave

nseidle, 2018-07-11:
You can compile from the [source](https://github.com/sparkfun/OpenLog/tree/master/firmware/OpenLog_Firmware/OpenLog) (it's just an Arduino sketch with a lib or two). Also, I had the sketch handy so I compiled it.

[OpenLog.ino.with_bootloader.eightanaloginputs.zip](https://github.com/sparkfun/OpenLog/files/2186352/OpenLog.ino.with_bootloader.eightanaloginputs.zip)

This version might be slightly non-production: I was doing some work on the other issue (SPI current consumption).

BigDave365, 2018-07-16:
I got it sorted...what an adventure.

Thanks for the help. I did end up using your .Hex file, but the root of the problem was I was using a chip from bare metal and did not program the fuses. The defaults will give the results I had (no response to serial and garbage data).

For those that might need them, here is what worked for me with Avrdude and a UsbASP:

High = 0xD6
Low =  0xFF
Ex  = 0x05

Lock Bit = 0x0F

Thanks again for the project and help.

-Dave

closed 2018-07-16 by BigDave365

nseidle, 2018-07-16:
That is good news! Just so I understand, you have your own ATmega328P+microSD design/setup? Not one of our OpenLogs? I ask because I'm always trying to make our issue system better. Next time I'll be sure to specifically ask what a user's setup is.