sparkfun/OpenLog #221: echo and verbose commands issue

danistrigaro, 2018-07-31:
### Subject of the issue
My OpenLog [1] seems to stuck after setting the echo or verbose parameter to off/on. It is not able to receive any more commands. 

### Your workbench
* I'm using Arduino Mega wired to this OpenLog [1]
* The firmware version is the 4.0
* wiring schema ---> RX of the Openlog to TX2 of the Arduino; TX of the Openlog to RX2 of the Arduino
* The OpenLog is powered by 5V; the Arduino by 7V 
* With other version of OpenLog everything works fine
* if I do not use the verbose or echo commands the others (set, md, read, cd, etc.) work as expected

### Steps to reproduce
Type through Serial2 "echo off/on" or "verbose off/on" before any other commands 

### Expected behavior
Parameters configured without blocking the next commands sent.

### Actual behavior
The echo or verbose command works as expected and a '>' will appear, but for any other commands sent after, the exception "unknown command " is prompted together with a mirrored question mark.
Once the Arduino is reset a series of "1212121121212..." is printed in the serial monitor

Best
Daniele

[1] https://www.epal.pk/product/openlog-serial-to-micro-sd-data-logger-module/

nseidle, 2018-08-01:
Ick. Sounds like the unit is crashing when verbose is disabled. I'll try to dig into it today. In the mean time are you able to operate with verbose on?

danistrigaro, 2018-08-02:
Thank you for your response! There is the same behavior if we put verbose or echo to on...

nseidle, 2018-08-02:
It sounds like the verbose and echo commands are causing problems. If you do a [emergency reset](https://learn.sparkfun.com/tutorials/openlog-hookup-guide#troubleshooting), does the OpenLog record data as expected?

> With other version of OpenLog everything works fine

Do you mean you have another OpenLog board that works ok or did you load a different firmware version?

Since we did not build this board it can be difficult to troubleshoot. Please contact EPal and see if they have support.

danistrigaro, 2018-08-20:
I solved the issue by buying a new OpenLog directly from Sparkfun.
I tried to flash the firmware using Arduino UNO as ISP but it did not work.
I wrote to epal (linking also to this discussion) but they said that I have to update the firmware.
Thanks for you kind and fast support!

nseidle, 2018-08-20:
Thanks for coming back to let us know! I'm sorry Epal is shipping units with different firmware. If you have any other problems please ask.

closed 2018-08-20 by nseidle