sparkfun/OpenLog #224: Verbose and echo mode always turn on.

usernamesarelreadytaken, 2018-11-21:
### Verbose and echo mode always turn on.
I try to change config of openlog using config.txt I can change baud rate normally but for echo and verbose, I can't turn them off.

### How I connect openlog to my data streaming device
- I connect Tx, Rx directly to my Rx, Tx of my device.

### Steps to reproduce
I write my config.txt as 
> 115200,26,3,0,0,0,1
> baud,escape,esc#,mode,verb,echo,ignoreRX

When I use my openlog, it always show my config as I wrote in config.txt

but when I format my SDcard and try to see current config of my openlog it show as
> 115200,26,3,0,1,1,1
> baud,escape,esc#,mode,verb,echo,ignoreRX

If I change baud rate to 57600 by changing config.txt to 
> 57600,26,3,0,0,0,1
> baud,escape,esc#,mode,verb,echo,ignoreRX
and checking my current config on openlog, it show as 
> 57600,26,3,0,1,1,1
> baud,escape,esc#,mode,verb,echo,ignoreRX

Please help

nseidle, 2018-11-28:
I thought this [sounded familiar](https://github.com/sparkfun/OpenLog/issues/208). Looks like I changed how the verbose setting was stored, whereas the echo setting was stored the original way. Not good. Let me dig into it.

nseidle, 2018-11-28:
Thanks for reporting this bug! I've committed a new INO. Please load [this new INO](https://github.com/sparkfun/OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino) and let me know if you still have the issue.

closed 2018-11-28 by nseidle

usernamesarelreadytaken, 2018-11-29:
@nseidle Thanks. I have try it. No issue found now.

nseidle, 2018-11-29:
Very good! Thanks for testing it out.