sparkfun/OpenLog #225: Feature: Flow Control

rsaxvc, 2018-12-28:
I'd like to float the idea of a configuration option to turn the TX-pin into a CTS pin. This involve turning the OpenLog into an RX-only device, and probably require disabling the escape sequence. This could enable even faster baud rates.

nseidle, 2018-12-30:
I can see how it would be possible without too much work. My main concern is the lack of space. Would a separate, or side, version be of interest to you?  Sort of like the current 'light' version?

I agree, higher baud rates would be possible but I doubt we'd get much higher bandwidth. Is that something you need specifically?

rsaxvc, 2019-01-07:
I think a separate build would be fine if it doesn't fit easily in the normal build.

Specifically, I'm after higher baudrates, not higher bandwidth - my UART is bit-banged, so the slower the baudrate the more CPU cycles per byte, but I could easily put the CPU to sleep waiting for an interrupt on CTS.