sparkfun/OpenLog #226: Checking if OpenLog is connected

Keudra, 2019-01-03:
I want to check first if OpenLog is conected to my board before I start sending informations.
My idea was to create a new file and delete it afterwards to see if it works. Is there a better way to do it?

I'm using a stm32L100 with I2C for communication.

nseidle, 2019-01-03:
I'm slightly confused. Are you using [OpenLog](https://github.com/sparkfun/OpenLog) (serial interface) or [Qwiic OpenLog](https://github.com/sparkfunX/Qwiic_OpenLog) (I2C)?

Regular Serial OpenLog: You can check to see if the '12<' characters come through

Qwiic I2C OpenLog: You can check to see if the I2C device responds to the correct I2C address.

Keudra, 2019-01-04:
I'm using the serial interface with UART. I2C was an other project sorry for the confusion. 
I would like to check if it's avalible without checking a visual interface. I'm planning to use it by its own using some selfmade functions.

Keudra, 2019-01-09:
I'm try to fetch the answer from the OpenLog but it seams that it isn't working, because I ounly get the same data from the UART. It doesn't mather if I send a command befor it or not, it's always 536887084.
This is the function how I would like to see if the OpenLog is connected to the board.
`HAL_StatusTypeDef checkLogger(){
       char buffer[100];
	int l;

	l = sprintf(buffer,"new test.txt\r\n");

	HAL_UART_Transmit(&huart1, buffer, l, 1000);

        HAL_UART_Receive(&huart1, buffer, 1, 1000);
	if(strcmp(buffer, 536887084) == 0){	//thats the part where I would check if it received <
                l = sprintf(buffer,"rm test.txt\r\n");

		HAL_UART_Transmit(&huart1, buffer, l, 1000);
		return HAL_OK;
         }
         return HAL_Error;
}
`

closed 2019-01-14 by Keudra