sparkfun/OpenLog #231: Openlog LS command question

danistrigaro, 2019-04-12:
Dear all,

I am using OpenLog as datalogger to store environmental data. I do not understand which is the order of the files returned by the 'ls' command. At the beginning I thought that it was by alphabetic order, then I thought it was by creation time, but now I noticed that still is not like this.

Could you please give me some information about it?

Thank you very much in advance

Regards

Daniele

nseidle, 2019-04-16:
OpenLog is based on [SdFat](https://github.com/greiman/SdFat) written by Bill Greiman and inherits its ls command. The ls function is defined in the docs as:

[image]

ls function is defined [here](https://github.com/greiman/SdFat/blob/master/src/FatLib/FatFilePrint.cpp#L99). It is not apparent to me what the listing format is. It may be something lower level such as the order in which the files were written to the SD card.

loricrotser, 2019-10-28:
Daniele, did you need further help with this issue? If you do, please feel free to look into our [forums](https://forum.sparkfun.com/index.php) - we have tech support folks who can get you going in the right direction. Thank you!

closed 2019-10-28 by loricrotser