sparkfun/OpenLog #232: Empty Log File

rzurbuch, 2019-10-29:
### Subject of the issue
I have the OpenLog connected to some proprietary hardware.  Our HW is sending UART data (115200bps) to the OpenLog.  We also have the OpenLog RXI line connected to an o-scope to verify UART traffic.  When we power up the HW and the OpenLog, we see activity on the o-scope for a bit and it stops (let's call this Activity#1).  There is around 100 bytes of data sent to the OpenLog before going idle. We remove the SDCard and the log is empty.  We plug the SDCard back into the OpenLog.  We make the HW send some more UART data (maybe 100 more bytes, let's call this Activity #2).  We remove the SDCard and we see data but it is from Activity #1.  For some reason the buffer isn't being purged.  My understanding is that the buffer should be sent to the SDCard when there is 500ms of UART idle time.

### Your workbench
* What platform are you using?  OpenLog v4.0, Current boot mode: New file
 
* What version of the device are you using? I'm not sure where this information is located.  We bought from DigiKey and haven't changed the code on the OpenLog.  Is there a firmware version?  v4.0
* How is the device wired to your platform?  GRN (DTR): open, RXI connected to UART at 3.3V, TX0:  open, VCC = 5V
* How is everything being powered?  Our Proprietary HW is powering the unit with 5V.
* Are there any additional details that may help us help you?
     From the "disk" command:
     Card type: SDHC
     Manufacturer ID: 9F
     OEM ID: TI
     Product: SD32G
     Version: 6.1
     Serial number: 671776948
     Manufacturing date: 10/2018
     Card Size: 31104 MB

### Steps to reproduce
I can't seem to reproduce on a standalone OpenLog.  I'm only able to reproduce when it is connected to our HW.

### Expected behavior
I expect that all UART data should be recorded to the SDCard.  It shouldn't sit in the RAM of the micro.

### Actual behavior
See above.

rzurbuch, 2019-10-30:
> Test #1:  Show that the log file in the SD Card is empty after sending data to the OpenLog ...
1.  Power up the OpenLog
2. wait for the OpenLog to go to sleep (>2seconds)
3. Send the OpenLog some UART data:  "This is the 1st message."
4. If you look at the current log in the SDCard, it should be empty.

Test #2:  Show that the log file doesn't contain all the data sent to the OpenLog.  The last bit of data is stored in the OpenLog buffer and isn't sent to the SD Card.
1.  Power up the Open Log.
2.  wait for the OpenLog to go to sleep (>2seconds).  I used 3 seconds.
3.  Send the OpenLog some UART data:  "This is the 1st message."
4.  wait for the OpenLog to go to sleep (>2seconds).  I used 3 seconds.
5.  Send the OpenLog some UART data:  "This is the 2nd message."
6.  If you look at the current log in the SDCard, it should show the first message ("This is the 1st message" but won't show the second message ("This is the 2nd message.").

nevercast, 2020-01-15:
I think I may be seeing this issue also, on a fresh OpenLog bought from SparkFun