sparkfun/OpenLog #235: SparkFun_Qwiic_Relay library 1.2.0 warnings when -fpermissive is not set

PickyBiker, 2020-08-03:
The Quad Solid State library Example2_Quad_Relay_Basics sketch compiles without errors or warnings using the Arduino UNO in the Arduino IDE. That is because with that board in that IDE the -fpermissive compile flag is set by default.

If you try to compile that board using a Teensy board, the -fpermissive flag is not set and the compiler generates the numerous warnings. Please correct the code so  it works without warnings when using Teensy boards.

In addition, the default I2C address is "RELAY_ADDR 0x6D // Alternate address 0x6C" but when an I2C scan is performed the actual board address (just purchased and unaltered) appears to be 0x08.

Here is the environment being used:
OS: Windows 10
IDE: Arduino version 1.8.12
Boards: Arduino UNO and Teensy 4.0
SparkFun_Qwiic_Relay library 1.2.0
The Quad Relay board is the one in this kit: [](https://www.sparkfun.com/products/16833)

The errors are cause by 2 lines of code in the SparkFun_Qwiic_Relay.cpp file. The changes I tested (shown below) resolve the warnings issue but you may want to apply a different fix.

line #340   original _i2cPort->requestFrom(_address, 1);  
line #340   no warnings with:  _i2cPort->requestFrom((uint16_t)_address, 1);  

line #353   original:   _i2cPort->requestFrom(_address, 2); 
line #353  no warnings with:  _i2cPort->requestFrom(**(uint16_t)**_address, 2);

The requestFrom function expects the address argument to be a 16bit int but_address is an 8 bit value. Without -fpermissive this generates the warnings. 

The warnings are in this attached file.
[Errors.txt](https://github.com/sparkfun/OpenLog/files/5019243/Errors.txt)

PaulZC, 2020-08-04:
Hi @PickyBiker ,
You have opened this issue in the SparkFun OpenLog repo. Can you please close and re-open it in the Qwiic Relay library repo?
https://github.com/sparkfun/SparkFun_Qwiic_Relay_Arduino_Library
Best wsihes,
Paul

PickyBiker, 2020-08-04:
Closed because it was in the wrong forum.

closed 2020-08-04 by PickyBiker