sparkfun/OpenLog #238: write offset <LF> and <CR> inverted on micro SD

JKPina, 2020-11-10:
### write offset inverse chars
There is a different behaviour when writing normally to openlog and when writing with offset

### Your workbench
* I'm working with a MKR1000 and openlog v4 on Serial3 @9600 baud
* config.txt => 9600,36,3,0,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

### Steps to reproduce
This is my write offset function. I have simplified to a better understanding
`void guardar_registro_con_offset(String reg) {
  openlog_command_mode();
  Serial3.println("read LOGTIMER.csv");
  openlog_while_not('\r');  // catch echo
  
  int byte_offset = -2;  
  int fila_found_in = 0;

  for (int timeOut = 0 ; timeOut < 100 ; timeOut++) {
    while (Serial3.available()) {
      String aux = Serial3.readStringUntil('\r');
      byte_offset++;
      if (aux.indexOf(reg_hora) > 0) {
        Serial3.print("write LOGTIMER.csv ");
        Serial3.print(byte_offset * 27);
        Serial3.write(13);

      openlog_while_not('\r'); 
      openlog_while_not('<');

      Serial3.println(reg);
      Serial3.print("\r");  // 13, CR

      timeOut = 100;
      break;
      }
    }
    delay(1);
  }

  sprintf(buff, "append LOGTIMER.csv\r");
  Serial3.print(buff);
  openlog_while_not('<');
}`

And this is how it looks in the microSD
[image]

### LF and CR inverted! (with the same script i can get proper CR and LF order)

9600, 2020-11-10:
Hello, you called?

JKPina, 2020-12-07:
Hi guys! Do anyone solve this issue?
Have a great week!