sparkfun/OpenLog #244: Feature Request: Add ability to 'rotate' separate log files of max size

EmbeddedMan, 2021-09-17:
OpenLog is *almost* perfect for our needs. The thing we need that it is missing is the ability to automatically start a new file after the log file reaches some maximum size, and then the ability to rotate through a list of log file names so that the older log files are replaced by newer ones.

This would give OpenLog the ability to continually log large amounts of data without losing it if the system is powered down, with a way to track what is the oldest data and what is newer data, to have a maximum size for the log files, and to have a maximum number of known file names and sizes on the SD card.

This functionality has been added and a pull request is forthcoming.

EmbeddedMan, 2021-09-17:
The biggest part of this change is the addition of a MODE_ROTATE to the existing three modes (MODE_NEWLOG, MODE_SEQLOG and MODE_COMMAND). Almost all of the other changes are guarded by this new mode. This means that this new behavior will only affect OpenLog if the new mode is used.

Two new config parameters are also defined. One is the Max Rotate File Size and the second is the Max Rotate Files (count).

Max Rotate File Size is a value from 1 to 255 representing the maximum allowed size of a log file when in MODE_ROTATE as expressed in MB. Defaults to 100MB.

Max Rotate Files is a value from 1 to 255 representing the maximum number of allowed log files when in MODE_ROTATE before the oldest one is over-written. Defaults to 100 files.

Both of these new configuration parameters are stored in EEPROM, as well as being read from the CONFIG.TXT file on boot if present. The default config string is now:

 "115200,103,214,0,1,1,0,100,100\0" //= 115200 bps, escape char of ASCII(103), 214 times, new log mode, verbose on, echo on, ignore RX false, 100MB max rotate file size, 100 max rotate files

The menu has been updated to show the new MODE_ROTATE option and print out the current filenumber.

EmbeddedMan, 2021-09-17:
Since the SparkFun OpenLog Hookup Guide is by far the most complete set of instructions/documentation for this software, it would need to be updated to include the additional features discussed in this issue. I'm more than happy to help edit/write those changes if there were a simple way for me to do so.

EmbeddedMan, 2021-09-17:
Right now we've only integrated this change into the OpenLog.ino firmware file, not the OpenLog_Minimal.ino or OpenLog_Light.ino files. If there is any interested in updating those as well, please let me know and I can submit another PR.

nseidle, 2021-09-30:
Hi EmbeddedMan - I'm in a holding pattern on this and the stack issue until the PR is in.

No need for PRs on Light or Minimal as those generally have no or few 'options'.

EmbeddedMan, 2021-09-30:
Understood. Things got crazy at work and I won't have the PR up for another week or so. Thanks for keeping track of this-