sparkfun/OpenLog #245: Unable to return to System Config menu after executing some of the 'set' menu commands (hang)

EmbeddedMan, 2021-09-18:
### Subject of the issue
Unable to return to System Config menu after executing some of the 'set' menu commands (hang)

### Your workbench
Windows 10 Pro, fresh compile of OpenLog firmware with Arduino IDE v1.8.15

Using genuine OpenLog hardware from SparkFun, purchased several months ago.

Using OpenLog.ino v4.2 (most recent at this date)

OpenLog hardware only connection is through FTDI cable to PC using jumper wires as described in SparkFun OpenLog hookup guide.

Powered only through FTDI power

Using TeraTerm on PC to communicate with OpenLog

Contents of config.txt file:
115200,26,3,0,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

No other files on SD card at boot

### Steps to reproduce
1) Connect to OpenLog from TeraTerm
2) Get "12<" as output from OpenLog
3) Hit <ctrl-z> three times rapidly
4) OpenLog drops down into command mode
5) Get "~>" from OpenLog
6) Type "set" then enter
7) See System Configuration menu
8) Type "1", do not press enter
9) Receive "1New file logging" from OpenLog
10) OpenLog is now hung. No combination of commands, characters, <ctrl-z>s will induce OpenLog to respond in any way. No log file is generated or appended. OpenLog must be rebooted (disconnect/reconnect serial connection) in order to regain any functionality.

### Expected behavior
10) OpenLog should return to normal command mode and accept commands like "?" and "set". At least that's what a code review shows should be happening.

### Additional Debugging Information
Adding printf()s in the code shows that recordConfigFile() is completely executing, saving the 'set' menu choice properly to the config file and EEPROM, and then returns to systemMenu() just fine, which returns back up into the while(1) loop in commandShell() just fine. But something crashes before another loop through the while(1) can be made and so we never get to the  NewSerial.print(F(">")); statement at the top of the loop. I'm stuck here - maybe a stack problem? Maybe some sort of timing related hang when printing serial?

I'm assuming this worked at some point back in maybe November 2018 when @nseidle made the last code change. Maybe it's something to do with the compiler settings of the newer Arduino IDEs?

EmbeddedMan, 2021-09-18:
Another note here: if you simply exit from the 'set' menu with the 'x' command, it works fine. You get the ">" prompt and all System Configuration menu options work fine. Menu option "4" also appears to work fine. But all the rest ("1", "2", "3", "5" and "6") cause the crash to happen.

EmbeddedMan, 2021-09-18:
I have confirmed that the line "if (readLine(buffer, sizeof(buffer)) < 1)" is where the crash is occurring. 

I have confirmed that commenting out the call to recordConfigFile() prevents the crash from occurring.

So something in recordConfigFile() is causing a problem that then makes the crash happen at the top of the commandShell() loop when readLine() is called.

EmbeddedMan, 2021-09-18:
I have confirmed that the exact same code, compiled under Arduino IDE 1.6.8 works fine. So something about the newer compiler/IDE/build process is what is causing the change in behavior. Maybe part of it is that the newer IDE/compiler (1.8.15) makes a much smaller image (25,188 bytes compared to 27,652 bytes for IDE v1.6.8).

It would still be great to figure out why readLine() is crashing. But at least we now know why it was working in the past and how to get it to work as a workaround (use older IDE version).

EmbeddedMan, 2021-09-18:
And it is the memset() in readLine() that is triggering the crash.

EmbeddedMan, 2021-09-19:
Confirmed: Stack corruption. In readLine(), the value of 'buffer' used in memset() is being changed by something in recordConfigFile(). The memory location 'buffer' points to the first time through the command loop (when things work) is 2238. But after a call to recordConfigFile() it has gotten changed to 190.

After an immense amount of debugging, I'm confident that there is some stack corruption going on inside recordConfigFile() which only shows itself as a crash when the memset() line in readLine() is called.

The workaround solution that I've found is to make the buffer[] array in commandShell() static. This appears to completely eliminate the problem, although it's not clear why. Another, more complicated way to eliminate the problem is to create a global volatile pointer to the buffer[] array, copy the address of the buffer[] array in commandShell(). This changes the way the compiler uses the global[] array in such a way that the crash doesn't happen. Again, I'm not exactly sure how.

Is this worth a pull request? Or is it better to just use an older Arduino IDE? Not sure. I'll leave this hear for anyone who wants to dive deeper into this problem, but I'm not able to spend any more time on it at present.

nseidle, 2021-09-20:
Hi EmbeddedMan - Thanks for digging into all this! If I following the issue correctly, when you compile and load the 'stock' firmware, you have this issue. But if you use an off-the-shelf unit from SparkFun (with a binary that has been presumably compiled with an earlier version of Arduino) the issue goes away? If that's true, I may need to de-emphasize this issue a bit. I haven't touched this firmware in a long time and OpenLog sells very well so I'm hesitant to risk adding a potential variable to the stability of the firmware.

That said, your feature request is a great one. I'll try to make time for this issue and the feature in the coming weeks.

EmbeddedMan, 2021-09-20:
I don't know about the stock firmware from SparkFun - I did not test this feature on any of the 20 something OpenLogs we ordered before I put my own customized firmware on (see upcoming PR for adding log rotate mode). I know that compiling the latest code from this repo with the latest Arduino IDE version does cause the failure. You probably should double check a unit from stock just to be sure it doesn't have this problem. My assumption is that it's fine, as nobody else has complained about it.

Agreed with this not being a big issue for production OpenLogs if your existing hex file is working fine. However, maybe you should create a 'production' branch of the repo with an updated Readme saying which version of IDE you used to compile the production hex file so that people can get to that version if they want. If your production code isn't going to change, but the code here will continue to add features/bugfixes, that should probably be notated somewhere obvious (The OpenLog hookup guide maybe?) I just don't want users to get confused by thinking that the latest code from this repo with the latest IDE version gives them the same thing that comes from SparkFun.

EmbeddedMan, 2021-10-12:
Heh, that's good at least. Now we just need to find a solution (or strongly warn users away from using newer IDE versions). The engineer in me really wants to know why this is failing and find a way to fix it. :-)

EmbeddedMan, 2021-10-12:
Yeah at this point what we could use is somebody who has experience debugging Arduino code with a hardware debugger. I suspect that if you had that ability and a little time it would become obvious what's getting corrupted and how. Just too hard to do with printf() debugging only.

nseidle, 2021-10-12:
Found it. It looks like I hit this problem before. [This line's comments](https://github.com/sparkfun/OpenLog/blob/master/firmware/OpenLog_Firmware/OpenLog/OpenLog.ino#L996) points out that **strcpy_P()** adds a null pointer but that strlen does not consider \0 as a character. So the array needs +1. I suspect `strcpy_P(configFileName, PSTR(CFG_FILENAME));` was dutifully copying the 'config.txt' string, then adding \0 past the last spot of configFileName[] causing corruption to the RAM next door. In your case, it was buffer[30]. By casting `static char buffer[30];` you pushed the corruption to a different part of RAM, but it was still getting corrupt. 

In your PR, I've made sure both config.txt and SEQLOG00.TXT are terminated and any use of strcpy_P has the appropriate sized array to store the incoming string.