sparkfun/OpenLog #249: Random data corruption

creagan01, 2022-05-03:
### Subject of the issue
This is more of a question than a issue.  Just wondering if this is something anyone may have seen before.

I am seeing random corrupted data samples, about every 2000 or 3000 samples.

Each sample is a data structure containing a 4 byte timestamp and a 4 byte sample value.
The timestamp is never corrupted only the sample value.
The corrupted values are very similar (first 3 bytes always the same) but way out of the range of what the sample value should look like.
Sample rate is 65Hz.
Data snippet shown below is a test example logging an incrementing 32bit integer.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/creagan/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/creagan/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

39597 | 2178
-- | --
39613 | 2179
39629 | 2180
39644 | 1.48E+20
39660 | 2182
39676 | 2183
39692 | 2184
39707 | 2185

</body>

</html>

### Your workbench
* What platform are you using?
 SparkFun OpenLog wired to a Noridic nRF52840
* What version of the device are you using? Is there a firmware version?
 OpenLog v4.2
* How is the device wired to your platform?
RX/TX to UART?  It is a prototype so wiring is a bit messy but it works, well, almost.  :)
* How is everything being powered?
The SparkFun board is powered by the same 3.3v that the Nordic device is using, so, some dropout there.
* Are there any additional details that may help us help you?
Running at 115,200 baud, using command mode.  I escape to command mode about every 4.5hrs  and start appending to a new log file but all that seems to be working fine.

Below is the config.txt:
115200,90,8,2,0,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

I can provide further information if needed.

Thanks

creagan01, 2022-05-05:
After further debugging, I have found the problem with my code.  This issue can be closed.

nseidle, 2022-05-05:
Good find! Thanks for letting us know.

closed 2022-05-05 by nseidle